YosysHQ/arachne-pnr #4: Assigning IO to pin 78 gives assert

jamesbowman, 2015-07-24:
Changing the rot.v example's assignments in `rot.pcf` from 

    set_io D5 95

to

    set_io D5 78

then running `make` gives:

    arachne-pnr: src/vector.hh:70: BasedVector<T, B>::reference BasedVector<T, B>::operator[](BasedVector<T, B>::size_type) [with T = int; typename std::vector<_RealType>::size_type B = 1ul; BasedVector<T, B>::reference = int&; BasedVector<T, B>::size_type = long unsigned int]: Assertion `i >= B && i < B + v.size()' failed.

cseed, 2015-07-25:
This works for me.  Are you using the current master branch?  Do you have the seed of the failing run?

jamesbowman, 2015-07-25:
Aha, thanks. Synced, and it no longer asserts.
Closing.

closed 2015-07-25 by jamesbowman