YosysHQ/arachne-pnr #10: Feature request: Ignore unused constraints

wd5gnr, 2015-08-16:
It would be nice to have an option to allow unused entries in a PCF file to be ignored. If there is a way to do that, I haven't found it.

cseed, 2015-08-16:
My only hesitation is that if you use this option but misspell a constraint, the signal will be given a random pin assignment.  Depending on your setup, that could destroy your device.  Are you sure this is what you want?  If so, I'll add it.

wd5gnr, 2015-08-16:
Hmm... a lot of commercial tools offer this as an option (but keep the error as the default). However, it seems to me you have the same problem if you synthesis tool doesn't force you to use `default_nettype none right? If I have a constraint called FOO and I type FOOx by mistake, you'll merrily assign an unused and unconstrained IOB to it with the same result. Or do you protect against that somehow?

I suppose an alternative would be to allow a PCF file and keep the current behavior, but also allow a "board configuration" that the PCF file would override and having extras in it would be ok (but not in the PCF). So you could have:

--board-default-constraints=icestick

Or whatever and that would define a bunch of constraints. Of course, then you have to publish the names of the constraints and not everyone is going to like them so that makes a whole other set of problems.

If I were you, I'd add the option with a stern warning if you find unused constraints, but not a fatal error. So with no option it would behave like it does now. With the option it would change the fatal error to something like:

***** IMPORTANT WARNING: Constraint x not used. If you think your code uses this pin, then it is probably misspelled and assigned to a random pin. Programming this bitfile could create an output that will short circuit or otherwise be misconfigured and your chip may be damaged. 

Or something like that.

cseed, 2015-08-17:
OK, I will add the option and generate a warning.

I don't understand the case that can fail.  If you have a constraint FOO but misspell the signal FOOx, you get an error.  Alternatively, if you misspell the constraint FOOx with signal FOO, you also get an error.

wd5gnr, 2015-08-17:
My use case is I have a PCF file for the iCEStick that defines all the pins. The way it is now, I have to go comment out the ones I do not use or else I get an error.

cliffordwolf, 2015-08-17:
Just a suggestion: Add support for a PCF syntax like `set_io -noerr D1 99` to disable error reporting for individual statements.

cseed, 2015-08-19:
I like Clifford's suggestion.  It seems to have all the right properties.  I checked in the change, syntax is `set_io --warn-no-port ...'.

closed 2015-08-19 by cseed

cliffordwolf, 2015-08-19:
My intuition to `--warn-no-part` would be that this enables warning about missing ports (warn when there is no port). I'd suggest something `--no-warn-port` or `--no-warn-no-port` instead.

cseed, 2015-08-19:
That's exactly what it does: warn (instead of giving an error).  But I guess the latter is the more salient feature.  How about `-no-err-port-missing`?

cliffordwolf, 2015-08-19:
> That's exactly what it does: warn

Ah! Ok. Sorry. Now I get it... Yes, the name is fine of course.

wd5gnr, 2015-08-19:
Thanks to you both, both for the new option and the whole thing in general.