YosysHQ/arachne-pnr #11: Error trying to process blockram

guanix, 2015-08-24:
I'm trying to create a very simple example of using blockram on an iCE40. My code is [in this gist](https://gist.github.com/guanix/86e1dcaa0e16cd867270). yosys runs fine, but arachne-pnr gives this output:

    seed: 1
    device: 1k
    read_chipdb +/share/arachne-pnr/chipdb-1k.bin...
      supported packages: tq144
    read_blif ramtest.blif...
    prune...
    read_pcf ramtest.pcf...
    instantiate_io...
    D1
    0x7fac02e1b190
    RDATA[0]
    clk
    0x0
    pack...

    After packing:
    IOs          2 / 96
    LCs          0 / 1280
      DFF        0
      CARRY      0
      CARRY, DFF 0
      DFF PASS   0
      CARRY PASS 0
    BRAMs        1 / 16
    WARMBOOTs    0 / 1
    GBs          0 / 8

    promote_globals...
      promoted 0 nets
      0 globals
    realize_constants...
      realized 0
    place...
      initial wire length = 23
      final wire length = 4

    After placement:
    PIOs       2 / 96
    PLBs       1 / 160
    BRAMs      1 / 16

      place time 0.00s
    route...
      pass 1, 0 shared.
      route time 0.01s
    write_txt ramtest.txt...
    libc++abi.dylib: terminating with uncaught exception of type std::out_of_range: map::at:  key not found
    make: *** [ramtest.bin] Abort trap: 6

You can also see that the .txt file generated appears truncated.

Forgive me if it's a problem with the Verilog file. I know that this isn't fully correct use of the blockram, but I wanted to isolate the problem. If there's a working example of blockram use out there, I would love to see it.

I'm at arachne-pnr d820f31fc3 and yosys eb38722e98.

cseed, 2015-08-25:
Thanks for the detailed bug report.  Fixed!  https://github.com/cseed/arachne-pnr/commit/7052728dd03b8959616e3e3206cdc85f048c158f

closed 2015-08-25 by cseed

qaarah, 2018-07-24:
Hi,
I used a blif file to input Arachne-pnr for place and route. I got outputs that are .asc and txt format. Can I get .blif file from Arachne after Place and route, Just like VPR? Is there any possibility to get a blif file from arachne-pnr after place and route.
Thanks

daveshah1, 2018-07-24:
You can use `--post-place-blif` to get a post-placement BLIF (as you would have seen if you used `-h`). Post-routing BLIFs are not supported.

In the future please open a new GitHub issue, or use an alternative support method such as one of the IRC channels (##openfpga or #yosys), rather than posting comments on random old and unrelated issues.

qaarah, 2018-07-24:
Yes, That I used and got a post placed blif file. 
Sure, 
Thanks