YosysHQ/arachne-pnr #12: adding icestorm package variant results in assertion failure

alibabashack, 2015-10-12:
Hi, I just tried to add bonding information for the vq100 package of the 1k chip in the icestorm project (see https://github.com/alibabashack/icestorm/tree/1k-vq100). When I try to build arachne-pnr with the resulting chip-db I get the following assertion failure. I'm unsure if my modifications in icestorm are wrong or this is a problem in arachne:

mkdir -p share/arachne-pnr
bin/arachne-pnr -d 1k -c /usr/local/share/icebox/chipdb-1k.txt --write-binary-chipdb share/arachne-pnr/chipdb-1k.bin
seed: 1
device: 1k
read_chipdb /usr/local/share/icebox/chipdb-1k.txt...
arachne-pnr: src/chipdb.cc:852: void ChipDB::finalize(): Assertion `tile_pos_cell[t][pos] == 0' failed.
make: *** [share/arachne-pnr/chipdb-1k.bin] Aborted (core dumped)

cseed, 2015-10-15:
I can confirm this is a problem in arachne.  I should have time in the next day or two to fix it.

alibabashack, 2015-10-15:
Thanks for your feedback!

The arachne, icestorm and yosis projects are absolutely amazing. I'm a huge fan. The academic world will at least produce a ton of interesting papers based on it, now that FPGAs can finally unfold their true potential. Thank you very much for your great work.

cseed, 2015-10-16:
Thanks for your patience!  Should be fixed:

```
$ arachne-pnr 
seed: 1
device: 1k
read_chipdb +/share/arachne-pnr/chipdb-1k.bin...
  supported packages: tq144, vq100
read_blif <stdin>...
^C
```

I tried a small example, but it crashed because there is a pin (91) at 6 17 0, but there is no ieren entry for 6 17 0.

closed 2015-10-16 by cseed

alibabashack, 2015-10-16:
Great, thanks. I'm going to investigate and test as soon as my hardware arrives.