YosysHQ/arachne-pnr #14: Make the --warn-no-port the default in the pcf

benallard, 2015-10-17:
Could we make the --warn-no-port the default when reading the constraint file ?

At the moment, when absent, it's considered a fatal error to define a port there, and not have in in the top level. I don't think this should be a fatal error, but I might be missing something there.

cseed, 2015-10-17:
The fatal error is intended to catch the case a constraint or port name is misspelled and a port is assigned a random pin by accident.  If the port is output and it is assigned to a pin driven by another device, for example, it could damage the hardware.  I think this is the behavior of most vendor tools.

Increasingly, I'm thinking if you're deploying on real hardware, all toplevel ports should require constraints (and then we can safely ignore extra, unused constraints).  If you're not deploying on hardware (e.g. simulation, testing) then it doesn't matter and we can relax the fatal error.  Thoughts?  You might take a look at the previous discussion: https://github.com/cseed/arachne-pnr/issues/10.

benallard, 2015-10-21:
I do have the exact same use case as for #10: I have one big pcf file that took me some time to write, and I'd like to be able to use that pcf again and again without having to comment out the lines I don't use.

Beside, I don't really follow how an unused pin would catch a typo more efficiently than it would not: If I made a typo somewhere, this message will likely catch it, or not.

That being said, it remains your decision how you want to go on with that.

Thanks so much for the work so far !

cliffordwolf, 2015-11-01:
I have thought a lot about this in the last two weeks and have a suggestion:

Currently a pin without a placement constraint is assigned a pin by arachne-pnr automatically, and an unused  placement constraint produces an error.

I would suggest to keep the current behavior when *no* PCF file is passed with `-p`, but use a different behavior when a PCF file is present:

- Produce an error when a PIN has no placement constraint
- Produce a warning when a placement constraint is unused

I would further suggest to add the `set_io` option `--no-warn` that makes arachne=pnr suppress the warning message. This is useful for "board default PCF files".

This will still produce an error when a pin name is misspelled (either in the PCF or Verilog file), but will work fine when a PIN is not used in the Verilog design.

benallard, 2016-01-26:
I wasn't aware of the fact that the pcf file was optional. That clears a bit indeed.

I agree with your proposal @cliffordwolf, it makes perfect sense.

How should we progress ? I would gladly dedicate cycles to that issue if I get some pointers on where, or how to start.

cliffordwolf, 2018-01-02:
I've now implemented this in 24f6b9c.

closed 2018-01-02 by cliffordwolf