YosysHQ/arachne-pnr #20: global network conflict PLL vs GB

zeldin, 2015-12-12:
When using PLL:s, arachne-pnr does not correctly handle conflicts with GB:s during placing.

Using the attached BLIF as input, arachne-pnr places a GB (for the promoted net `$auto$rtlil.cc:1705:NotGate$12488$2`) at (0 16), and the PLL at (16 0), causing them both to drive `glb_netwk_6`.

(Device 8k)
[bad.blif.txt](https://github.com/cseed/arachne-pnr/files/60113/bad.blif.txt)
[bad.pcf.txt](https://github.com/cseed/arachne-pnr/files/60114/bad.pcf.txt)

cseed, 2015-12-17:
@zeldin Sorry I haven't responded to this sooner.  I've been quite busy at work.  Thanks for the concise bug report.  I can reproduce it and will have a fix soon, certainly by this weekend.

cseed, 2015-12-25:
Fixed (finally!): https://github.com/cseed/arachne-pnr/commit/1a4fdf96a7fd08806c032d41a2443c8e17c72c80

closed 2015-12-25 by cseed