YosysHQ/arachne-pnr #21: Add support for the blif construct .subckt

SebastianBoe, 2015-12-20:
Hi,

from what I see in the blif parser, it appears that the .subckt 
construct is not supported yet. 

I would like this supported because Yosys generated .subckt 
commands when the -noflatten option was used.

Symptom:
top.blif:42: fatal error: unknown directive

Workaround:
Don't use subckt. (For the yosys flow, don't use the -noflatten option)

PS: Thanks for creating this project. Since there exists a workaround, it might make sense
to close this as won't fix.

cseed, 2015-12-25:
This would be reasonable to add.  It might also be useful for non-yosys tools that want to generate .blif files directly.  I'm going to leave it open as an enhancement.  Thanks for the suggestion!

porglezomp, 2019-04-26:
I ran into this issue while working with yosys.
Another workaround that might be helpful to people who aren't happy with avoiding `-noflatten` for any reason. Instead of writing:
```
synth_ice40 … -blif <name.blif>
```
you can flatten your design after synthesis and write it then.
```
synth_ice40 …
flatten
write_blif -gates -attr -param <name.blif>
```

whitequark, 2019-04-26:
FYI Arachne is essentially not maintained at this point as nextpnr is a far more capable replacement.