YosysHQ/arachne-pnr #23: Consider more permissive open source licensing

scanlime, 2016-01-31:
Hello! I'm working on an open source library which will be generating FPGA bitstreams dynamically, using yosys + icestorm + arachne-pnr. If you'd like more info about what I'm doing, here's the site: https://github.com/scanlime/wiggleport

I was wondering if you would consider re-releasing arachne-pnr under a more permissive license that would allow it to be integrated into other software packages? Something similar to the MIT or ISC license would be ideal, but even LGPL would be a step forward.

Thank you for your consideration!

cliffordwolf, 2016-07-13:
I second that.

Especially in non-Linux embedded applications (running on OSes like Contiki or BitThunder) it might be necessary to link everything in one binary. GPL can be a showstopper in such cases.

drom, 2016-11-23:
I am building my JavaScript based UI, and using ASM.JS version YosysJS. MIT would be perfect for that.

closed 2017-01-17 by cseed