YosysHQ/arachne-pnr #24: Inferring SB_IO with D_IN_0 from $_TBUF_ cells

cliffordwolf, 2016-02-02:
I've been playing with inference of SB_IO from `$_TBUF_` cells and encountered a few problems. Afaics it boils down to the following test case:

```
module test(input din, en, output dout, inout io);
  assign io = en ? din : 1'bz, dout = io;
endmodule
```

```
yosys -p 'synth_ice40 -blif tristate.blif' tristate.v
arachne-pnr -o tristate.asc tristate.blif
```

Which should of course be equivalent to:

```
module test(input din, en, output dout, inout io);
  SB_IO #(
    .PIN_TYPE(6'b 1010_01),
    .PULLUP(1'b 0)
  ) iobuf (
    .PACKAGE_PIN(io),
    .OUTPUT_ENABLE(en),
    .D_OUT_0(din),
    .D_IN_0(dout)
  );
endmodule
```

But arachne-pnr produces the following error for BLIF generated from the first Verilog code:

```
tristate.blif:14: fatal error: toplevel inout port 'io' not connected to SB_IO PACKAGE_PIN
```

The BLIF generated by Yosys looks reasonable imo:

```
.model test
.inputs din en io
.outputs dout io
.names $false
.names $true
1
.names $undef
.gate $_TBUF_ A=din E=en Y=io
.attr src "tristate.v:2"
.names io dout
1 1
.end
```

wolfmankurd, 2016-03-03:
I thought I might try to add an example.

ali@ali-debian:~/Code/Verilog/ice40/test$ arachne-pnr -d 1k -P qn84 -p cap.pcf cap.blif -o cap.asc
seed: 1
device: 1k
read_chipdb +/share/arachne-pnr/chipdb-1k.bin...
  supported packages: cb121, cb132, cb81, cm121, cm36, cm49, cm81, qn84, swg16tr, tq144, vq100
read_blif cap.blif...
cap.blif:18: fatal error: toplevel inout port 'BTN[0]' not connected to SB_IO PACKAGE_PIN

module cap (CLK_3P3_MHZ, LED, BTN);
   input CLK_3P3_MHZ;
   inout reg [3:0] BTN;
   output reg [3:0] LED;

   always@(posedge CLK_3P3_MHZ) begin
	LED = BTN;
   end
endmodule // cap

ali@ali-debian:~/Code/Verilog/ice40/test$ cat cap.blif 
# Generated by Yosys 0.6+1 (git sha1 c89f61c, clang 3.5.0-10 -fPIC -Os)

.model cap
.inputs CLK_3P3_MHZ BTN[0] BTN[1] BTN[2] BTN[3]
.outputs LED[0] LED[1] LED[2] LED[3] BTN[0] BTN[1] BTN[2] BTN[3]
.names $false
.names $true
1
.names $undef
.gate SB_DFF C=CLK_3P3_MHZ D=BTN[0] Q=LED[0]
.attr src "cap.v:6|/usr/local/bin/../share/yosys/ice40/cells_map.v:2"
.gate SB_DFF C=CLK_3P3_MHZ D=BTN[1] Q=LED[1]
.attr src "cap.v:6|/usr/local/bin/../share/yosys/ice40/cells_map.v:2"
.gate SB_DFF C=CLK_3P3_MHZ D=BTN[2] Q=LED[2]
.attr src "cap.v:6|/usr/local/bin/../share/yosys/ice40/cells_map.v:2"
.gate SB_DFF C=CLK_3P3_MHZ D=BTN[3] Q=LED[3]
.attr src "cap.v:6|/usr/local/bin/../share/yosys/ice40/cells_map.v:2"
.end

vpecanins, 2016-05-05:
Hi,

I am having the same issue with arachne-pnr and tristate buffers. Below is an example.v code:

```
module top (
        input  clk,
        inout LED1, LED2, LED3, LED4, LED5
);
        localparam LOG2DELAY = 22;
        reg [LOG2DELAY-1:0] counter = 0;
        reg [1:0] outcnt = 0;

        always@(posedge clk) begin
                if (counter == 0) begin
                        case (outcnt)
                          2'b00: outcnt <= 2'b01;
                          2'b01: outcnt <= 2'b10;
                          default: outcnt <= 2'b00;
                        endcase
                end
                counter <= counter + 1;
        end

        assign LED5 = outcnt[0] ? 1'bz : outcnt[1];
endmodule
```

I tried to compile this code with the `icestick.pcf` and `Makefile` files from the `examples/icestick` folder included in the source of icestorm. Yosys performs good and generates successfully `example.blif` but arachne-pnr shows the error:

```
arachne-pnr -d 1k -o example.asc -p icestick.pcf example.blif
seed: 1
device: 1k
read_chipdb +/share/arachne-pnr/chipdb-1k.bin...
  supported packages: cb121, cb132, cb81, cm121, cm36, cm49, cm81, qn84, swg16tr, tq144, vq100
read_blif example.blif...
example.blif:196: fatal error: toplevel inout port 'LED1' not connected to SB_IO PACKAGE_PIN
Makefile:11: recipe for target 'example.asc' failed
make: *** [example.asc] Error 1
```

cliffordwolf, 2016-05-05:
arachne-pnr is a bit picky regarding the top level port declarations. You should only declare a port as inout if your are actually using it as inout. In this case you are not using LED1..LED4 at all. So you can declare them as either input or output and everything will be fine. You are using LED5 as a tristate output, but do not use its value when you tristate it. So it is sufficient to declare it as output:

```
module top (
        input  clk,
        output LED1, LED2, LED3, LED4, LED5
);
        localparam LOG2DELAY = 22;
        reg [LOG2DELAY-1:0] counter = 0;
        reg [1:0] outcnt = 0;

        always@(posedge clk) begin
                if (counter == 0) begin
                        case (outcnt)
                          2'b00: outcnt <= 2'b01;
                          2'b01: outcnt <= 2'b10;
                          default: outcnt <= 2'b00;
                        endcase
                end
                counter <= counter + 1;
        end

        assign LED5 = outcnt[0] ? 1'bz : outcnt[1];
endmodule
```

When you are actually using LED5 as tristate inout, then the safest way to do that right now is by manually instantiating an SB_IO cell. See for example `raspi_io` in this code: https://github.com/cliffordwolf/icotools/blob/master/icosoc/common/icosoc_raspif.v

vpecanins, 2016-05-05:
Thanks, manually instantiating the SB_IO effectively solves the problem. Here is a dummy example program with bidirectional port. It compiles well with the mentioned Makefile. 

```
module top (
        input  clk,
        output LED1, LED2, LED3, LED4,
        inout LED5
);
        localparam LOG2DELAY = 22;
        reg [LOG2DELAY-1:0] counter = 0;
        reg [1:0] outcnt = 0;
        reg dummy;

        always@(posedge clk) begin
                if (counter == 0) begin
                        case (outcnt)
                          2'b00: outcnt <= 2'b01;
                          2'b01: outcnt <= 2'b10;
                          default: outcnt <= 2'b00;
                        endcase
                end
                counter <= counter + 1;
        end
        
        assign LED1 = dummy;

        SB_IO #(
          .PIN_TYPE(6'b 1010_01),
          .PULLUP(1'b 0)
        ) led_io (
          .PACKAGE_PIN(LED5),
          .OUTPUT_ENABLE(outcnt[0]),
          .D_OUT_0(outcnt[1]),
          .D_IN_0(dummy)
        );
endmodule
```

cseed, 2016-07-12:
@cliffordwolf I have one problem with your original example: the generated $_TBUF_ drives two toplevel ports.  My original intention was to treat explicitly instantiated SB_IO and $_TBUF_ similarly, and I only allow them to drive a single toplevel port.

cseed, 2016-07-12:
Disregard the mistakenly sent previous message.  I have a plan for all of this now.

PR #110 (2018-05-12): Map inout ports to input/output if they're not actually tristate
Fixes #24.

This doesn't handle the degenerate case of doing e.g.
```verilog
module top(inout x[7:0]);
  assign x[0] = x[1];
endmodule
```
and I think can't possibly handle it. In any case it does not seem important to me to handle this case implicitly.
commit 22e2c6ab (2018-05-12): Map inout ports to input/output if they're not actually tristate.

Fixes #24.
closed 2018-05-12 by cliffordwolf