YosysHQ/arachne-pnr #26: Fails to build when downloaded as .zip

klammerj, 2016-02-13:
The git commands in the makefile don't work then.
I'm getting:
```
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
Not a git repository
To compare two paths outside a working tree:
usage: git diff [--no-index] <path> <path>
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
Not a git repository
To compare two paths outside a working tree:
usage: git diff [--no-index] <path> <path>
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
echo "const char *version_str = \"arachne-pnr 0.1+0+0 (git sha1 , g++ `g++ --version | tr ' ()' '\n' | grep '^[0-9]' | head -n1` -O0 -fno-inline)\";" > src/version_02097.cc
```
It does not stop there, but continues.
It finally fails with:
```
g++ -Isrc -std=c++11 -MD -O0 -fno-inline -g -Wall -Wshadow -Wsign-compare -Werror  -o bin/arachne-pnr src/arachne-pnr.o src/netlist.o src/blif.o src/pack.o src/place.o src/util.o src/io.o src/route.o src/chipdb.o src/location.o src/configuration.o src/line_parser.o src/pcf.o src/global.o src/constant.o src/designstate.o src/version_02097.o -lm
make: *** No rule to make target '/usr/share/icebox/chipdb-1k.txt', needed by 'share/arachne-pnr/chipdb-1k.bin'.  Stop.
```

Now I'm not sure... will git clone fix it or am I missing some textfiles?

kevinmehall, 2016-02-13:
You need to install [IceStorm](https://github.com/cliffordwolf/icestorm) for the chipdb files.

klammerj, 2016-02-13:
OK. just got that figured out aswell... the git thing is valid, thoug..

PR #57 (2017-01-20): Fix version info bug.
Fixes #26
closed 2017-01-20 by cseed