YosysHQ/arachne-pnr #28: [Windows] Error when reading the chipdb*.bin in Windows

Obijuan, 2016-03-01:
I've cross-compiled arachne for working in windows. Everything works ok except that it cannot read the chipdb*.bin files.

This is the error I get:

```
arachne-pnr -d 1k -p t1.pcf t1.blif -o t1.asc
seed: 1
device: 1k
read_chipdb +/share/arachne-pnr/chipdb-1k.bin...
fatal error:std::istream::read: No error 
```

Initially I though maybe the problem was in the paths: it could not find the chipdb-1k.bin file. So, I copy the chipdb-1k.bin into the same folder than arachne and tested it againg: 

```
arachne-pnr -d 1k -p t1.pcf -c chipdb-1k.bin t1.blif -o t1.asc
seed: 1
device: 1k
read_chipdb chipdb-1k.bin...
fatal error:std::istream::read: No error 
```

I tried to use the chipdb-1k.txt instead of chipdb-1k.bin and it worked:

```
arachne-pnr -d 1k -p t1.pcf -c chipdb-1k.txt t1.blif -o t1.asc
seed: 1
device: 1k
read_chipdb chipdb-1k.bin...
  supported packages: ......
  ....
```

I've checked that the generated chipdb-1k.bin files are ok (tested on linux). I've also generated the chipdb-1k.bin from Windows. But the error still remains.

Obijuan, 2016-03-01:
Debugging the code, the fatal error is generated in the function: ChipDB::bread(ibstream &ibs) in the file chipdb.cc.  

More specific, the appers when this statement es executed:

ibs >> device
      >> width
      >> height
    // n_tiles = width * height
      >> n_nets
    // n_global_nets = 8
      >> packages
      >> loc_pin_glb_num
      >> iolatch
      >> ieren
      >> extra_bits
      >> gbufin;
      >> tile_colbuf_tile  //-- The fatal error is produced here
      >> tile_type
   // net_tile_name
      >> net_names
      >> tile_nets_idx // tile_nets
      >> tile_nonrouting_cbits
      >> n_cells
      >> cell_type
      >> cell_location
      >> cell_mfvs
      >> cell_type_cells
    // bank_cells
      >> switches
    // in_switches, out_switches
      >> tile_cbits_block_size;

Commenting the different fields and testing, I've seen that the fatal error is produced when the field tile_colbuf_tile is processed.

Any clues on what could be the problem? 

I am using the i686-w64-mingw32-g++ for cross-compilation  (version 4.9.2)

benpye, 2016-03-16:
Looking briefly (I'm looking to do this myself), chipdb is just a binary serialization and includes some STL types. `tile_colbuf_tile` is actually an `std::map` and so will be incompatible, you'll need to generate the bin on Windows.

benpye, 2016-03-20:
I just read your message again, and tried this locally. I was wrong evidentially about it being a cross platform issue. I believe that writing an `std::map` to and from a file like that is actually not supported, and works on Posix platforms by coincidence, there is no guarantee that the implementation can be written to and from a file (may contain pointers etc). It seems the solution would be to use some sort of serialisation, this would slow down reading from the file somewhat depending upon what was used. http://www.boost.org/doc/libs/1_60_0/libs/serialization/doc/index.html is pretty well known and should be easy to use, though it isn't the fastest (see https://github.com/thekvs/cpp-serializers). Cereal also looks good and seems to have support for all stl types so I may try and use that. This would break compatibility with existing chipdb files, but I don't think that's really an issue?

Obijuan, 2016-03-21:
Great! If you need any help just let me know ;-)

benpye, 2016-03-21:
@Obijuan I have fixed this, see #29

Obijuan, 2016-03-29:
Great! I am testing it know. I will tell you more when I finish

Obijuan, 2016-03-29:
I can confirm that it works :-)  Thank a lot!  :-D :-D

PR #29 (2016-03-29): Fix chipdb read failure on Windows
Windows will convert `\n` to `\r\n` and vice versa on read/write in text mode, binary mode must be stated explicitly to prevent this. This fixes #28
closed 2016-03-29 by cseed