YosysHQ/arachne-pnr #32: arachne-pnr: Unknown Model: $_DLATCH_P_

vpecanins, 2016-05-06:
Hello mates,

I am trying to compile an Asynchronous FIFO for the ICE40. It is used to share data between 2 buses with different clock frequencies. The [model](http://www.asic-world.com/examples/verilog/asyn_fifo.html) is taken from the page World of ASIC. It has been copied&pasted and compiled with the commercial tool Ice Cube 2 and it works.

I am trying to do the same with the open source toolchain. However, `arachne-pnr` fails with the following error:

`[...] fatal error: unknown model '$_DLATCH_P_'`

I assume the problem comes when trying to place the D LATCH described below:

```
//'Status' latch logic:
    always @ (Set_Status, Rst_Status, Clear_in) //D Latch w/ Asynchronous Clear & Preset.
        if (Rst_Status | Clear_in)
            Status = 0;  //Going 'Empty'.
        else if (Set_Status)
            Status = 1;  //Going 'Full'.
```

`yosys` effectively synthetizes this as a D LATCH and writes the following line to the `.blif` file:
```
.attr src "./aFifo.v:105|/usr/local/bin/../share/yosys/ice40/cells_map.v:18"
.gate $_DLATCH_P_ D=$abc$1111$n75 E=$abc$1111$n71 Q=afifo1.Status
```

Then, the problem must be that arachne-pnr does not know the model for the D latch. Any ideas?

Regards

cliffordwolf, 2016-05-06:
iCE40 has no hardware support for latches. The Lattice tools implement them by creating logic loops. (If you check the icecube build logs you will see that you get warnings that timing arcs are removed because of the logic loops created to implement latches.) There are so many reasons why you should never infer latches anyways.. Only one of them is that it is virtually impossible to verify them for timing. Most Verilog style guides simply forbid latches categorically.

Interestingly "World of ASIC" is the page where I got most of my test cases for really bad code.. :)

All that being said, I have now added support for that in Yosys commit https://github.com/cliffordwolf/yosys/commit/6fe3d5a1cf938081110db0470def2b2687dd665f. Update Yosys and your code will synthesize to a logic loop.

But be warned: icetime is not as forgiving as the lattice tools. It will simply report the loop as longest (infinite) path and give you a 0 MHz timing estimate.

cseed, 2016-07-10:
I'm marking this as closed since Clifford added support for latches in Yosys.

closed 2016-07-10 by cseed