YosysHQ/arachne-pnr #36: Apparently invalid bitstream for LVDS input

mo101, 2016-06-29:
According to http://www.clifford.at/icestorm/io_tile.html for a 1K chip:

> When an input pin pair is used as LVDS pair ... then the four bits IoCtrl IE_0/IE_1 and IoCtrl REN_0/REN_1 are all set, as well as the IoCtrl LVDS bit. 

However, it seems that the IE and REN bits are only set for one of the pins of the LVDS pair, not both, and as a result the LVDS input does not work reliably. Setting the bits manually makes the input work reliably.

For example, with the BLIF file lvds.blif:

```
.model top
.inputs in
.outputs out
.names $false
.names $true
1
.names $undef
.gate SB_IO D_IN_0=x PACKAGE_PIN=in
.attr src "lvds.v:8"
.param IO_STANDARD "SB_LVDS_INPUT"
.param PIN_TYPE 000001
.gate SB_IO D_OUT_0=x PACKAGE_PIN=out
.attr src "lvds.v:13"
.end
```
constraints file lvds.pcf:
```
set_io in 34
set_io out 98
```
and command line:
`arachne-pnr -d 1k -o lvds.asc -p lvds.pcf lvds.blif`
the result in io_tile 0 2 is:
```
.io_tile 0 2
000000000000000000
000100000000000001
000000000000000000
000000000000000001
000000000000000000
000000000000000000
000000000000000000
000000000000000000
001000000000000000
000100000000000000
000000000000000000
000000000000000000
000000000000000000
000000000000000000
000000000000000000
000000000000000000
```
B1[3] and B9[3] are set, but B6[2] and B6[3], which I think should also be set, are not.

cliffordwolf, 2016-06-30:
> and as a result the LVDS input does not work reliably

this sounds like you have a setup to test LVDS functionality in-hardware. I think I have fixed the issue now in commit 81aa60f. Please test and let me know if the issue is indeed fixed.

mo101, 2016-06-30:
The new commit generates an identical bitstream for that I/O tile to my modified version, and the LVDS input works reliably. Many thanks for the prompt response.

closed 2016-06-30 by mo101

bmentink, 2016-10-18:
Hi,

Can some one tell me how to modify the following verilog code to use LVDS inputs? I am not sure how the pins are defined ...

```
 SB_IO #(.PIN_TYPE(6'b0000_00)) _io (
        .PACKAGE_PIN(pin),
        .INPUT_CLK(clk),
        .D_IN_0(rd));
```
I apologise if I am asking in the wrong place ....

RGD2, 2016-10-18:
You will need an external differential resistor across the pin pairs, and
only IO bank3 supports lvds input.
See lattice Technical Note TN1253.

I'm doing this myself: I just use a TI DS90LV012A chip to translate each
LVDS to a single ended 3.3V CMOS logic. Those don't need any external
resistors, and they're guaranteed good up to 200 MHz, and may do 250MHz.

There's also a companion ds90lv011a which is an LVDS transmitter.

The ice40hx8 breakout board only supports a couple of possible lvds input
pairs, and has no space for the necessary resistors for receiving or
sending.

bmentink, 2016-10-18:
@RGD2 

Thanks for that. Do you know what pins are associated with bank 3, I could not find that info anywhere in the Lattice docs. Also, how do I know what other pin is associated with the pin I select?
I guess these are selected by the yosys tool.

RGD2, 2016-10-18:
You can download the pin specs as a spreadsheet from lattice.
Which bank they're in is shown for each package.

bmentink, 2016-10-18:
Yep, had seen that. There pin designations don't tie up with the breakout boards ones .... apart from the C1/C2 type designations. I found the most useful guide the breakout schematics, from that there seems to be only 3 differential inputs available on bank3 ... guess that will have to do.

My intention is to turn the LVDS input pins into ADC pins for an Electronic Drum Brain machine ...... 
I have all the code to implement that including filters etc ...