YosysHQ/arachne-pnr #39: Support for primitive placement constraints

vpecanins, 2016-07-08:
Hi mates,

I am working on a FPGA project involving generating an internal clock using only a Ring Oscillator (the same as used by Clifford Wolf in his pressure-sensing project [here](http://svn.clifford.at/handicraft/2015/ringosc/)

My issue is that, as the frequency of such oscillator depends on the propagation delay of the signals, it varies when the design is routed differently. It would be nice to be able to assign a position manually to the involved LUT's, so the frequency is preserved when other parts of the design are changed.

In the commercial IceCube utility from Lattice, this can be done by specifying the placement of the primitives in a PCF file (generated with the 'floor planner' graphical tool)

The first lines specify a "name" to the desired LUTs in the design:
`
###BLE List 48
ble_pack ring_osc_0 {ring_osc1.buffers[0]}
ble_pack ring_osc_1 {ring_osc1.buffers[1]}
ble_pack ring_osc_2 {ring_osc1.buffers[2]}
ble_pack ring_osc_3 {ring_osc1.buffers[3]}
ble_pack ring_osc_4 {ring_osc1.buffers[4]}
ble_pack ring_osc_5 {ring_osc1.buffers[5]}
ble_pack ring_osc_6 {ring_osc1.buffers[6]}
ble_pack ring_osc_7 {ring_osc1.buffers[7]}
`

Then, the LUTs are packed on groups of 8, and the position of each group is specified:
`
clb_pack ring_osc_clb_0 {ring_osc_0,ring_osc_1,ring_osc_2,ring_osc_3,ring_osc_4,ring_osc_5,ring_osc_6,ring_osc_7}
set_location ring_osc_clb_0 8 5
`
The first line packs the LUTs in the same CLB, the second line assigns the position of the CLB in the tile (8, 5)

Any ideas on how to implement this on arachne-pnr? Maybe the file pcf.cc could be modified to parse these lines, then assign the positions to the affected elements and tell the optimizer not to move them.

cliffordwolf, 2016-07-08:
I have on my todo list adding support to something like a `(* loc="X Y Z" *)` attribute on cells (X, Y as tile coordinate, Z the position within the tile), but I haven't found the time so far to look into it. Such an attribute would also allow people to write placers outside of arachne-pnr and only use arachne-pnr for routing and creating the configuration data.

Afaict there are three components to this: (1) adding support for reading the placement information, via loc attribute or additional statements in the .pcf file, (2) adding something to Placer::place_initial() that uses this information to place the cells in question and (3) add something to the placer that allows cells to be locked in place (if something like that doesn't exist already).

phanrahan, 2016-07-12:
I posted a similar request to the reddit list, and recently started to try to do this.

I added an option to --place-only. And I changed the --route-only version to read in the standard loc attribute (instead of the cell).

The current version still fails the tests, but it is starting to come together. Feel free to comment and make suggestions. It's a complicated system.

https://github.com/phanrahan/arachne-pnr

drom, 2017-01-20:
Placement constrains would nice option. I am experimenting with datapath generation tool and semi-custom placement would be great feature to have.

cseed, 2017-01-20:
@phanrahan Any progress on this?  I'm trying to set aside time to work on arachne-pnr again, but if you're still making progress on this I don't want to step on your toes.  There's plenty to do.

phanrahan, 2017-01-27:
Yes, some progress. I forked the repository (under github.com/phanrahan). I added basic placement constraints using LOC. This involved a bunch of changes to place.cc. You may want to review them and see if I am headed in the right direction. However, I wouldn't have much time to work on this in the near future. I agree with you, there is plenty of cool stuff to do.

madscientist159, 2017-04-01:
@phanrahan This feature would make certain things a lot easier on our side since arachne-pnr still has no support for timing constraints.  What's the current status, and what would be the best way for us to pick up development if you've abandoned it?

Thanks!