YosysHQ/arachne-pnr #45: invalid .names entry

tusj, 2016-09-29:
I am trying out the arachne-pnr on a blif file generated with yosys, and it fails on invalid .names entry.

I am using these versions:
`Yosys 0.6+209 (git sha1 9b8e06b, gcc 6.1.1 -march=x86-64 -mtune=generic -O2 -fstack-protector-strong -fPIC -Os)`
`arachne-pnr 0.1+171+0 (git sha1 52e69ed, g++ 6.2.1 -O2)`

The build script I am using with yosys is:
```
read_verilog verilog/Simple/counter.v
read_verilog verilog/Simple/Simple_a4.v
read_verilog verilog/Simple/Simple_binaryTo7Segment_0.v
read_verilog verilog/Simple/Simple_binaryTo7Segment.v
read_verilog verilog/Simple/Simple_BitPackBitVector1.v
read_verilog verilog/Simple/Simple_BitPackBitVector2.v
read_verilog verilog/Simple/Simple_bundler_sbundler.v
read_verilog verilog/Simple/Simple_countThe_scountThe.v
read_verilog verilog/Simple/Simple_debounce.v
read_verilog verilog/Simple/Simple_limiter.v
read_verilog verilog/Simple/Simple_released.v
read_verilog verilog/Simple/Simple_toggle.v
read_verilog verilog/Simple/Simple_topEntity.v

hierarchy -check -top counter

proc;    opt
memory;  opt
techmap; opt
flatten; opt

write_blif -unbu counter.blif
```

I added the flatten command because before that the generated blif file contained .subckt definitions which currently are not supported by arachne-pnr #21 .

I then invoke arachne-pnr like this:
`arachne-pnr -d 1k -p Go_Board_Constraints_arachne_pnr.pcf -P vq100 -o counter.asc counter.blif`

The error message I then get from arachne-pnr is:
`seed: 1
device: 1k
read_chipdb +/share/arachne-pnr/chipdb-1k.bin...
  supported packages: cb121, cb132, cb81, cm121, cm36, cm49, cm81, qn84, swg16tr, tq144, vq100
read_blif counter.blif...
counter.blif:11: fatal error: invalid .names entry
`

The first content (including the line of the error) of the blif file used is:

```
# Generated by Yosys 0.6+209 (git sha1 9b8e06b, gcc 6.1.1 -march=x86-64 -mtune=generic -O2 -fstack-protector-strong -fPIC -Os)

.model counter
.inputs i_Switch_1 system1000 system1000_rstn
.outputs o_LED_1 o_Segment1_A o_Segment1_B o_Segment1_C o_Segment1_D o_Segment1_E o_Segment1_F o_Segment1_G
.names $false
.names $true
1
.names $undef
.names Simple_topEntity_inst.Simple_binaryTo7Segment_0_result_0.map[6].Simple_BitPackBitVector1_0.result o_Segment1_A
0 1
```
[counter.blif.txt](https://github.com/cseed/arachne-pnr/files/500568/counter.blif.txt)
I should mention that I am a beginner in FPGAs.

I don't know whether this is a bug or me not using the tools correctly.

cliffordwolf, 2016-09-29:
You have to map your design to the iCE40 architecture in your Yosys script. You don't do that. Instead you generate a logic-level blif file like one would use as input to ABC.

Mapping to iCE40 cells is usually done using the command `synth_ice40`. See literally any IceStorm example or tutorial for a usage example.

In your case, instead of running

```
hierarchy -check -top counter

proc;    opt
memory;  opt
techmap; opt
flatten; opt

write_blif -unbu counter.blif
```

Simply run:

```
synth_ice40 -top counter -blif example.blif
```

If you don't want to use the `synth_ice40` command, see `help synth_ice40` for more information on how the command does what it does, and use that as a starting point for your custom script.

tusj, 2016-09-29:
Thank you so much, sorry for the unnecessary report! It is working like expected with the synth_ice40 command.

closed 2016-09-29 by tusj