YosysHQ/arachne-pnr #48: LUT interior expression

drom, 2016-11-23:
Currently the output produced by the tool provides the information about the LUT configuration as the number.
example: 
```json
  "type": "ICESTORM_LC",
  "parameters": {
    "LUT_INIT": 27030, <-- ???
    "CARRY_ENABLE": 1
  }
```
It would be beneficial (to visualization / navigation / extraction ... effort) to elaborate on LUT configuration in logical expression form.

cliffordwolf, 2016-11-23:
The number is the 16 bit truth table for the LUT (e.g. `27030 (dec)` is `0110100110010110 (bin)`). Since it is only a 4-input function you could probably easily create a [Karnaugh map](https://en.wikipedia.org/wiki/Karnaugh_map) from the LUT bits to get to a minimal sum-of-products expression from its minterms, or minimal product-of-sums expression from its maxterms.

Alternatively you could create a [BDD](https://en.wikipedia.org/wiki/Binary_decision_diagram) with the LUT bits on it leaves, reduce the BDD, and then display it as a circuit of multiplexers.

drom, 2016-11-23:
@cliffordwolf is the order is little endian? Is the following true?
```

27030 (dec)
0110100110010110 (bin)

 3    2    1    0  line
0110.1001.1001.0110

        i1    i0
       1  0  1  0
     
i2 0   0  1  1  0
   1   1  0  0  1
i3 0   1  0  0  1
   1   0  1  1  0
```

cliffordwolf, 2016-11-23:
This doesn't make much sense:

        i1    i0
       1  0  1  0

It should rather be something like:

       i0=1       i1=0
    i1=1 i1=0  i1=1 i1=0

I.e. the two left columns mean i0 is set, the two right columns mean that i0 is cleared.

And then for i1 there is an alternating pattern of columns.

Similar for i2 and i3 and the rows.

Regarding the bit ordering: see https://github.com/cliffordwolf/yosys/blob/master/techlibs/ice40/cells_sim.v#L121 for a model of the SB_LUT4 cell type. I3 selects if the upper or lower half or the truth table is used, I0 selects if an odd or even entry is used. the LSB of the truth table is selected if all inputs are zero.

drom, 2016-11-23:
@cliffordwolf I got it. Verilog prototype is better then a hundred words ;)

cseed, 2017-01-20:
Looks like @cliffordwolf answered this so I'm going to close it.

closed 2017-01-20 by cseed

drom, 2017-01-20:
Yes @cseed , @cliffordwolf answer works!