YosysHQ/arachne-pnr #50: Failing with "assertion failed" using SB_LVDS_INPUT

tcr, 2016-12-21:
I'm trying to get the following Verilog to compile, using SB_LVDS_INPUT. I can't find any examples online aside from the Lattice PDF, which I tried to follow, but am getting a failed assertion in arachne-pnr:

```
module Main (
    input clk,
    output LED1,
    output LED2,
    input PIO3_02,
    input PIO3_03,
);
  reg input_0;
  reg input_180;

  // Differential input, DDR data
  SB_IO #(
    .PIN_TYPE(6'b0000_00),
    .IO_STANDARD("SB_LVDS_INPUT")
  ) differential_input (
    .PACKAGE_PIN(PIO3_02),
    .LATCH_INPUT_VALUE ( ),
    .CLOCK_ENABLE ( ),
    .INPUT_CLK (clk),
    .OUTPUT_CLK ( ),
    .OUTPUT_ENABLE ( ),
    .D_OUT_0 ( ),
    .D_OUT_1 ( ),
    .D_IN_0 (input_0),
    .D_IN_1 ()
  );

    always @(*) begin
        LED1 <= input_0;
        LED2 <= input_180;
    end
endmodule
```

The error it is failing with:

```
Assertion failed: (valid(chipdb->cell_location[c].tile())), function place_initial, file src/place.cc, line 1104.
```

Relevant pcf for the HX1K ship I'm using:

```
set_io clk 21   # 12 MHz clock
set_io --warn-no-port LED1 99  # red
set_io --warn-no-port LED2 98  # red
set_io --warn-no-port PIO3_02 1 # PIO3_02/DP00A
set_io --warn-no-port PIO3_03 2 # PIO3_02/DP00B
```

I've investigated the place.cc file, it fails on this clause in `Placer::valid(int t)`:

```
if (g1) {
    ...
    Instance *inst1 = gates[g1];
    if (inst1->get_param("IO_STANDARD").as_string() == "SB_LVDS_INPUT")
        return false; // Fails here
```

I apologize if I'm overlooking something simple about how to set up a differential LVDS input on DP00A/B.

tcr, 2016-12-21:
When I switch `PACKAGE_PIN` to `PIO3_03`, layout succeeds. Is it the case I just have to use DPxxB instead of DPxxA for LVDS pairs?

cliffordwolf, 2018-03-04:
Fixed by #97.

closed 2018-03-04 by cliffordwolf