YosysHQ/arachne-pnr #54: Feature request: timing constraints

madscientist159, 2017-01-16:
Now that IceStorm has documented the timing for these devices (and in fact can generate a nice static timing report via icetime), would you consider adding timing-driven P&R to arachne?  We really need to be able to place a period constraint on a couple of our clock signals.

Thanks!

madscientist159, 2017-01-22:
@cseed Unfortunately we're really hitting a brick wall without the ability to constrain the master clock.  Is this something that's difficult to add?

Thanks!

cseed, 2017-01-22:
I don't think I'll get to it soon.  After writing arachne-pnr I took a new job and haven't much time to continue working on it.

I don't think it is too hard.  You need to load the timing database along with the chip database.  Then you need to add the timing information to the routing cost function.  I would probably route every signal timing-optimal (with conflicts) and iteratively reroute signals to resolve the conflicts without breaking the timing budget, similar to how the congestion aware router works now.  The hardest part will probably be balancing conflicts vs timing between multiple signals on a critical path.  I'd probably do some research to start, there might be standard approaches in the literature.  I'd welcome any contributions in this direction.  Sorry I don't have a better answer.

madscientist159, 2017-01-23:
@cseed Thank you for the update.  I can't promise much at this time, but if we did end up integrating an arachne-based workflow into a commercial product, would you be open to adding this feature under paid contract?

Thanks!

cseed, 2017-01-23:
@madscientist159, I'd certainly be open to this!  Please get in touch over email (cotton@alum.mit.edu) if you'd like to talk more.