YosysHQ/arachne-pnr #60: hx1k vq100 package not supported?

luc-tielen, 2017-02-12:
Hi,
I'm trying to use arachne to PNR a FPGA design for ice40-hx1k-vq100.
I get an error when I do the following command:
arachne-pnr -d 1k -P vq100 -p netlist.pcf top_module.blif -o top_module.asc

I found the list of options @ http://www.clifford.at/icestorm/ (see table).

This gives the following output (some output stripped to make it shorter):
...
device: 1k
read_chipdb +/share/arachne-pnr/chipdb-1k.bin...
  supported packages: tq144
fatal error: unknown package `vq100'
Makefile:6: recipe for target 'build' failed

Looks like the VQ100 package isn't supported anymore? Or is there something wrong with my installation?
(uname -a gives: Linux mint 4.4.0-62-generic #83-Ubuntu SMP Wed Jan 18 14:10:15 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux)

Thanks in advance.

cliffordwolf, 2017-02-12:
My guess would be that your version of arachne-pnr is built using a really old version of icestorm. With the current version of icestorm and arachne-pnr:

```
device: 1k
read_chipdb +/share/arachne-pnr/chipdb-1k.bin...
  supported packages: cb121, cb132, cb81, cm121, cm36, cm49, cm81, qn84, swg16tr, tq144, vq100
```

luc-tielen, 2017-02-12:
Thanks for the quick reply!
I build everything from source, seems to work now!

luc@mint ~ $ arachne-pnr 
seed: 1
device: 1k
read_chipdb +/share/arachne-pnr/chipdb-1k.bin...
  supported packages: cb121, cb132, cb81, cm121, cm36, cm49, cm81, qn84, swg16tr, tq144, vq100

The debian packages seem to be very outdated..
I will close the issue now :).

closed 2017-02-12 by luc-tielen