YosysHQ/arachne-pnr #66: map::at:  key not found

C-Elegans, 2017-06-21:
When trying to route a design using LUT instantiation, I get this error:
```
route...
libc++abi.dylib: terminating with uncaught exception of type std::out_of_range: map::at:  key not found
```

Does arachne-pnr not support cascading LUTs via the LO port of ICESTORM_LC? I've tried it with both I0 and I2, and it errors both times.

Test case:

test.v:
```verilog
module test (/*AUTOARG*/
   // Outputs
   LED,
   // Inputs
   clk, pin
   ) ;
   input clk, pin;
   output [1:0] LED;
   wire [1:0]  a;
   wire [1:0]  s;
   ICESTORM_LC #(
		 .LUT_INIT('b01),
		 .DFF_ENABLE(0))
   l1(
      .I0(pin), .LO(a[0]), .O(), .CLK(clk));
   ICESTORM_LC #(
		 .LUT_INIT('b01),
		 .DFF_ENABLE(1))
   l2(
      .I0(a[0]), .LO(a[1]), .O(s[0]), .CLK(clk));
    ICESTORM_LC #(
		 .LUT_INIT('b01),
		 .DFF_ENABLE(1))
   l3(
      .I0(a[1]), .LO( ), .O(s[1]), .CLK(clk));

   assign LED = s;
endmodule // test
```

test.pcf:
```
set_io LED[0] 99
set_io LED[1] 98
set_io pin 112
set_io clk 21
```

built with:
```
yosys -p "synth_ice40 -top test -blif test.blif" test.v
arachne-pnr -p test.pcf test.blif -o test.txt
```

C-Elegans, 2017-06-21:
With a similar but larger design, I have also gotten this error
```
route...
24506 -> 20635
Assertion failed: (unrouted.empty()), function route, file src/route.cc, line 749.
```

mmicko, 2017-11-26:
Adding next lines at line 161 of route.cc show the routing issue.

if (chipdb->tile_nets[t].find(tile_net_name) == chipdb->tile_nets[t].end())
{
   fatal(fmt("failed to rote:  " << p->name() << " to " << tile_net_name));
}

Thing is that lutff_7/lout does not exist in chipdb (it only exists for lutff_0 till lutff_6), and guess it that is how it is on real hardware as well, but would be good if @cliffordwolf  can confirm that is valid.

cliffordwolf, 2017-11-26:
> Adding next lines at line 161 of route.cc show the routing issue.

Can you create a PR for this?

> Thing is that lutff_7/lout does not exist in chipdb

Yes, this is correct afaik. The `lout` output of one lut can only be routed to `in_2` of the next lut in the tile. For `lutff_7` there is no next lut and therefore no `lout` signal. (When the FF is disabled then `lout` and `out` are the same. Otherwise `lout` is the signal before the FF.)

mmicko, 2017-11-26:
Sure, just made it now.