YosysHQ/arachne-pnr #78: Summary output report formats?

mithro, 2017-10-14:
Does Project IceStorm toolchain (yosys, arachne-pnr, IceStorm Tools) have some type of standard output format for getting reports about things like;

 * Messages like info, warnings & errors (for all stages)

 * Synthesis information like;
    * Parameters set for modules
    * Optimisations applied
    * Transformations
    * Component + Blackbox Statistics / Usage
    * etc

 * Place & Route like;
    * Resource usage
    * Timing information
    * etc

The reason I'm asking is that in my CI system, I would like to track how a design is changing over time. For example, I would like to understand if merging a change will dramatically increase resource usage, introduce new warnings, etc.

I don't quite know what would be the most useful to track yet, so also open to suggestions there :-). I don't really want the extreme low level details, more high level and hierarchical summaries.

I eventually hope to write some crappy Python scripts which translate the proprietary tools (like Vivado and ISE) into the Yosys / OpenFPGA formats to reuse the tracking for designs I can't yet use the open tools for.

Thank you for your help!

cliffordwolf, 2017-10-14:
Yosys uses special functions to log warnings and errors. But there is no API yet for writing them to an extra file. (By in `-q` mode only warnings and errors are printed to stdout.)

The Yosys command `stat` prints resource usage statistics. You can run something like `tee -o stat.txt stat` as last command in your Yosys script.

Practically all of the Yosys log is "Optimisations applied" and "Transformations". You can run a command with `trace` to print additional trace information for every design change caused by the command (for example `trace opt`).

qaarah, 2018-07-23:
Hi, 
I used a blif file to input Arachne-pnr for place and route. I got outputs that are .asc and txt format. Can I get .blif file from Arachne after Place and route, Just like VPR? Is there any possibility to get a blif file from arachne-pnr after place and route. 
Thanks

proppy, 2023-06-27:
note that `stat` now has a `-json` output: https://yosyshq.readthedocs.io/projects/yosys/en/latest/cmd/stat.html