YosysHQ/arachne-pnr #100: Obscure error message and core dump when PLL instance can't be routed

tomverbeure, 2018-03-18:
```	arachne-pnr -d 8k -P tq144:4k -p blackice-ii.pcf chip.blif -V post_place.v -o chip.txt```

The error message:
```
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
Aborted (core dumped)
```

The design mistake is that the input clock of a SB_PLL40_2F_PAD PLL is also used to clock a FF directly. Arachne should flag this as illegal.

[issue.zip](https://github.com/cseed/arachne-pnr/files/1822210/issue.zip)

daveshah1, 2018-04-01:
Thanks for reporting this. At some point it would be nice to produce better error messages for all placement and routing failures.

tomverbeure, 2018-04-13:
@ducky64,
I think your case is different than what I reported in that my case was a broken error warning, but the input design was indeed wrong.

In your case, the design seems to be correct yet it’s still bombing. That’s worse.

I suggest that you file a separate issue for this, so that it may get more attention (which it deserves.)

Tom

tomverbeure, 2018-05-22:
This issue is a dup of issue #77 and can be closed.

closed 2018-05-22 by tomverbeure