YosysHQ/arachne-pnr #103: ICE40HX8K-BG121

jkiv, 2018-04-01:
Good day,

I'm interested in using the [BG121](https://www.digikey.com/product-detail/en/lattice-semiconductor-corporation/ICE40HX8K-BG121/220-2148-ND/7784603) flavour of the ICE40HX8K, since it's a less expensive 0.8mm alternative to the CT256.

It's not mentioned in much of the Lattice documentation, but in some recent correspondence about the chip with a Lattice representative, he/she said "ICE40HX8K-BG121 package was just added in the datasheet last year so it's not one of the discontinued products."

Any chance of this package being supported?

daveshah1, 2018-04-01:
It looks like the BG121 is available in icecube without license control. I'm happy to add support for it to icestorm when I next have time. I agree this looks like a nice package option for some applications.

cliffordwolf, 2018-04-01:
@daveshah1 when you do that, can you quickly scan through the chip packages available in icecube and see if lattice added any other packages? Adding 1 new package or N new packages is probably about the same effort..

daveshah1, 2018-04-01:
Well, it looks like the BG121 has an identical pinout to the already supported CM121 package, based on both the pinout XLS and the pin to IO cell mapping in the floorplan viewer. It should just be a case of adding an alias in icestorm and a note in the docs.

daveshah1, 2018-04-02:
Added in cliffordwolf/icestorm#132. There don't seem to be any other packages that need to be added from icecube as far as I can see.

whitequark, 2018-05-14:
@cliffordwolf This can be closed.

closed 2018-05-14 by cliffordwolf