YosysHQ/arachne-pnr #123: UltraPlus 5K LVDS

widgetlords, 2018-07-20:
I am attempting to use pins 12 and 21 on the SG48 package as a LVDS input by instantiating SB_IO like this:
```
SB_IO #(
	.IO_STANDARD("SB_LVDS_INPUT"),
	.PIN_TYPE(6'b000000)
) SB_IO (
	.INPUT_CLK(sys_clk),
	.PACKAGE_PIN(lvds_pin),
	.D_IN_0(d_0),
	.D_IN_1(d_180)
);
```
where `lvds_pin` maps to pin 12 on the FPGA.

arachne-pnr comes back with this:
```
fatal error: pcf error: LVDS port `lvds_pin' not in bank 3
```

This FPGA doesn't have a bank 3. TN1253 from Lattice doesn't specifically mention any differences for the UltraPlus either. A quick look through the code and I found this: https://github.com/cseed/arachne-pnr/blob/e4a365715bca8e8da184a234a9381a1f25029965/src/pcf.cc#L314
Does this mean that LVDS support is not accounted for on UltraPlus models yet?

daveshah1, 2018-07-20:
I don't think I ever tested LVDS on the UltraPlus (somewhat concerning given I just ordered PCBs that need it...)

Based on Lattice's reference designs for the UP, LVDS can be used on any bank. So you should be able to comment out that check safely. If you have the ability to test, it would be great to try and report whether it works? Thanks!

tomverbeure, 2018-07-20:
Since nobody has tested LVDS IOs for UP models, I expect that things will still fail after commenting out that check, but maybe I'm too much of a pessimist. :-)

In any case, the commenting out is only for initial testing purposed, right? We don't want to remove the check for ICE40 versions.

widgetlords, 2018-07-20:
I just tried commenting out both checks and I got this:
```
arachne-pnr: src/place.cc:1208: void Placer::place_initial(): Assertion `valid(chipdb->cell_location[c].tile())' failed.
```

daveshah1, 2018-07-20:
Thanks for checking, I'll have a look.

daveshah1, 2018-07-20:
This is now fixed, was another validity check to remove plus a small config tweak, and I've pushed it to a branch here: https://github.com/daveshah1/arachne-pnr/tree/lvds-5k

The configuration bits for the relevant IO tile are now 1:1 identical to those from iCEcube2, which is all the testing I can do locally.

widgetlords, 2018-07-20:
I switched to using pin 21 instead of 12 as the package pin and it now synthesizes using your lvds-5k branch. I'm not sure that it's fully working but I am getting the behavior that I expect from it.

daveshah1, 2018-07-20:
Some of the Lattice documentation for the UltraPlus refers to them as differential comparators rather than LVDS, so maybe there's a performance difference - I'm not sure.