YosysHQ/arachne-pnr #129: failed to place HFOSC

olegendo, 2018-10-09:
Hi,

I'm trying to build some simple test logic for the smaller iCE40 devices that uses the built-in oscillator and I'm getting

  fatal error: failed to place HFOSC

for "-d 384 -P qn32" and "-d 1k -P qn84".
It's OK for "-d 5k -P sg48" though.

Am I missing something?

Cheers,
Oleg

Reproducer verilog:

```
module top (output clkout);

wire clk;

SB_HFOSC #(.CLKHF_DIV("0b10")) hfosc (
  .CLKHFPU(1'b1),
  .CLKHFEN(1'b1),
  .CLKHF(clk)
);

reg [7:0] counter;

always @(posedge clk) begin
  counter <= counter + 1;
end

assign clkout = counter[7];
endmodule

```

olegendo, 2018-10-09:
According to official documentation ...

> The iCE40TM family, specifically iCE40 UltraTM, iCE40 UltraLiteTM and iCE40 UltraPlusTM, features two on-chip oscillators.
> [...]
> Two oscillators are available to users:
> • SB_LFOSC – Low Frequency Oscillator
> • SB_HFOSC – High Frequency Oscillator with output divider
> 

Or do you mean, it's just not supported by YosysHQ/arachne-pnr tools (yet)?

daveshah1, 2018-10-09:
Neither the Ultra nor the UltraLite are supported at all by the tools. The 1k and 384 refer to LP/HX series ice40s

olegendo, 2018-10-09:
Ouch.  I totally confused all the iCE40 variants.  Thanks for the clarification and sorry for the noise.

closed 2018-10-09 by olegendo

mithro, 2018-10-09:
@olegendo - @awygle has been looking into support for the Ultra and UltraLite devices if I understand?

olegendo, 2018-10-10:
It'd be certainly great if there was support for Ultra and UltraLite. iCE5LP1K-SG48ITR looks like an appealing device for some designs.

In my particular case here, it doesn't matter though.  I totally goofed it.  I was expecting HFOSC to work on an iCE40-LP384-SG32, because I didn't realize that there is this significant difference between LP/HX, Ultra, UltraLite, UltraPlus ...  it'd be helpful if they just put everything on one website, not on 3, and did not scatter the information across a bazillion of documents ... oh well :)

kernelcolton, 2019-05-27:
So how does one program the ultralight if you can't use yosys?  is there another toolchain somewhere?

olegendo, 2019-05-27:
AFAIK support for iCE5 devices has been added to yosys and nextpnr upstream not long ago.

daveshah1, 2019-05-27:
Support for iCE5 (iCE40 Ultra) devices is in Yosys and nextpnr, and I believe this includes the internal oscillators. There is no support for the iCE40 UltraLite yet.