YosysHQ/arachne-pnr #131: Unable to use differential input on LP384

smunaut, 2018-10-25:
Attached test case : 

[bug.tar.gz](https://github.com/YosysHQ/arachne-pnr/files/2513873/bug.tar.gz)

If I try to instanciate a LVDS input and connect the package pin to the 'A' side like this : 

```
	SB_IO #(
		.PIN_TYPE(6'b000000),
		.PULLUP(1'b0),
		.NEG_TRIGGER(1'b0),
		.IO_STANDARD("SB_LVDS_INPUT")
	) pm_cmt_iob_I (
		.PACKAGE_PIN(diff_p),
		.LATCH_INPUT_VALUE(),
		.CLOCK_ENABLE(1'b1),
		.INPUT_CLK(1'b0),
		.OUTPUT_CLK(1'b0),
		.OUTPUT_ENABLE(1'b0),
		.D_OUT_0(1'b0),
		.D_OUT_1(1'b0),
		.D_IN_0(out),
		.D_IN_1()
	);
```

Then I get this error : 

> place_constraints...
> fatal error: pcf error: LVDS port `diff_p' not a DPxxB input

And if I try to conect it to the 'B' side like the error says it should be done : 

```
	SB_IO #(
		.PIN_TYPE(6'b000000),
		.PULLUP(1'b0),
		.NEG_TRIGGER(1'b0),
		.IO_STANDARD("SB_LVDS_INPUT")
	) pm_cmt_iob_I (
		.PACKAGE_PIN(diff_n),
		.LATCH_INPUT_VALUE(),
		.CLOCK_ENABLE(1'b1),
		.INPUT_CLK(1'b0),
		.OUTPUT_CLK(1'b0),
		.OUTPUT_ENABLE(1'b0),
		.D_OUT_0(1'b0),
		.D_OUT_1(1'b0),
		.D_IN_0(out),
		.D_IN_1()
	);
```

Then I get this error : 

> place...
> arachne-pnr: src/place.cc:1208: void Placer::place_initial(): Assertion `valid(chipdb->cell_location[c].tile())' failed.

smunaut, 2018-10-25:
Mmm, apparently it's because I have in the .pcf and the .v the companion pad is 'present' (even though not connected to anything).
I had done this for clarity to but I guess it triggers a bug because a SB_IO is auto added at some point and it's not valid.

davidthings, 2018-12-06:
@smunaut I am having the same exact problem, but I don't quite understand your fix. 

I want a differential input, not DDR.

```
module top (
    input CLK,     // 16MHz clock
    input  PIN_7,  // DAT_IN_P (IOL_10A)
//    input  PIN_9,  // DAT_IN_N (IOL_10B)
    output PIN_15,
);

    wire DAT_IN;

    SB_IO #(.PIN_TYPE( 6'b000001 ), .IO_STANDARD( "SB_LVDS_INPUT") ) dat_in (
        .PACKAGE_PIN(PIN_7),
        .D_IN_0 (DAT_IN)
    );

    assign PIN_15 = DAT_IN;

endmodule
```
With .pcf

```
set_io --warn-no-port PIN_7 D1
# set_io --warn-no-port PIN_9 E1
set_io --warn-no-port PIN_15 D9

# 16MHz clock
set_io --warn-no-port CLK B2 # input
```

For this I get:

```
fatal error: pcf error: LVDS port `PIN_7' not a DPxxB input
```

When I reverse things completely (commenting out the PIN_7, reinstating PIN_9, etc.) it does seem work.

Does this just mean that we need to specify the B side of the pair?  So simple?

davidthings, 2018-12-06:
Thanks for the confirmation.

closed 2020-06-25 by smunaut