YosysHQ/arachne-pnr #133: fatal error: failed to route - pass 200, 1 shared.

gojimmypi, 2019-01-16:
I've been working with the [RISCV example project on TinyFPGA BX](https://discourse.tinyfpga.com/t/riscv-example-project-on-tinyfpga-bx/451) and have documented my experiences [here](https://gojimmypi.blogspot.com/2019/01/risc-v-on-fpga-tinyfpga-via-wsl.html). 

One the the things that really tripped me up was the `fatal error: failed to route` message with arachne-pnr. (full details shown below). 

I just wanted to document here that the resolution to this problem for me was to use the `-r` option to "randomize the seed". Specifically like this:

`arachne-pnr -r -d 8k -P cm81 -o hardware.asc -p hardware.pcf hardware.blif
`

Hopefully having this explicit GitHub issue will help others avoid the problems I had with this.

Here's the output from arachne-pnr _without_ the `-r`, showing the error.
```
$ /usr/local/bin/arachne-pnr -d 8k -P cm81 -o hardware.asc -p hardware.pcf hardware.blif
seed: 1
device: 8k
read_chipdb +/share/arachne-pnr/chipdb-8k.bin...
  supported packages: bg121, bg121:4k, cb132, cb132:4k, cm121, cm121:4k, cm225, cm225:4k, cm81, cm81:4k, ct256, tq144:4k
read_blif hardware.blif...
prune...
read_pcf hardware.pcf...
hardware.pcf:43: warning: no port `pin_3' in top-level module `hardware', constraint ignored.
hardware.pcf:44: warning: no port `pin_4' in top-level module `hardware', constraint ignored.
hardware.pcf:45: warning: no port `pin_5' in top-level module `hardware', constraint ignored.
hardware.pcf:46: warning: no port `pin_6' in top-level module `hardware', constraint ignored.
hardware.pcf:47: warning: no port `pin_7' in top-level module `hardware', constraint ignored.
hardware.pcf:48: warning: no port `pin_8' in top-level module `hardware', constraint ignored.
hardware.pcf:49: warning: no port `pin_9' in top-level module `hardware', constraint ignored.
hardware.pcf:50: warning: no port `pin_10' in top-level module `hardware', constraint ignored.
hardware.pcf:51: warning: no port `pin_11' in top-level module `hardware', constraint ignored.
hardware.pcf:52: warning: no port `pin_12' in top-level module `hardware', constraint ignored.
hardware.pcf:53: warning: no port `pin_13' in top-level module `hardware', constraint ignored.
hardware.pcf:56: warning: no port `pin_14' in top-level module `hardware', constraint ignored.
hardware.pcf:57: warning: no port `pin_15' in top-level module `hardware', constraint ignored.
hardware.pcf:58: warning: no port `pin_16' in top-level module `hardware', constraint ignored.
hardware.pcf:59: warning: no port `pin_17' in top-level module `hardware', constraint ignored.
hardware.pcf:60: warning: no port `pin_18' in top-level module `hardware', constraint ignored.
hardware.pcf:61: warning: no port `pin_19' in top-level module `hardware', constraint ignored.
hardware.pcf:62: warning: no port `pin_20' in top-level module `hardware', constraint ignored.
hardware.pcf:63: warning: no port `pin_21' in top-level module `hardware', constraint ignored.
hardware.pcf:64: warning: no port `pin_22' in top-level module `hardware', constraint ignored.
hardware.pcf:65: warning: no port `pin_23' in top-level module `hardware', constraint ignored.
hardware.pcf:66: warning: no port `pin_24' in top-level module `hardware', constraint ignored.
hardware.pcf:77: warning: no port `pin_25' in top-level module `hardware', constraint ignored.
hardware.pcf:78: warning: no port `pin_26' in top-level module `hardware', constraint ignored.
hardware.pcf:79: warning: no port `pin_27' in top-level module `hardware', constraint ignored.
hardware.pcf:80: warning: no port `pin_28' in top-level module `hardware', constraint ignored.
hardware.pcf:81: warning: no port `pin_29' in top-level module `hardware', constraint ignored.
hardware.pcf:82: warning: no port `pin_30' in top-level module `hardware', constraint ignored.
hardware.pcf:83: warning: no port `pin_31' in top-level module `hardware', constraint ignored.
instantiate_io...
pack...

After packing:
IOs          13 / 63
GBs          0 / 8
  GB_IOs     0 / 8
LCs          5769 / 7680
  DFF        1491
  CARRY      821
  CARRY, DFF 157
  DFF PASS   522
  CARRY PASS 63
BRAMs        20 / 32
WARMBOOTs    0 / 1
PLLs         0 / 1

place_constraints...
promote_globals...
  promoted clk_16mhz$2, 1688 / 1688
  promoted $abc$60698$n19, 555 / 555
  promoted resetn, 326 / 326
  promoted $abc$60698$n4557, 96 / 96
  promoted soc.cpu.pcpi_div.start, 163 / 163
  promoted soc.cpu.pcpi_mul.mul_waiting, 216 / 216
  promoted $abc$60698$n4366, 73 / 74
  promoted $abc$60698$n4562, 64 / 64
  promoted 8 nets
    3 sr/we
    4 cen/wclke
    1 clk
  8 globals
    3 sr/we
    4 cen/wclke
    1 clk
realize_constants...
  realized 1
place...
  initial wire length = 116235
  at iteration #50: temp = 12.8802, wire length = 104649
  at iteration #100: temp = 7.32627, wire length = 77712
  at iteration #150: temp = 3.76088, wire length = 52415
  at iteration #200: temp = 1.65526, wire length = 36991
  at iteration #250: temp = 0.0517603, wire length = 27723
  at iteration #300: temp = 9.23435e-07, wire length = 27303
  final wire length = 27268

After placement:
PIOs       18 / 63
PLBs       874 / 960
BRAMs      20 / 32

  place time 44.80s
route...
  pass 1, 407 shared.
  pass 2, 222 shared.
  pass 3, 150 shared.
  pass 4, 94 shared.
  pass 5, 65 shared.
  pass 6, 66 shared.
  pass 7, 58 shared.
  pass 8, 39 shared.
  pass 9, 47 shared.
  pass 10, 38 shared.
  pass 11, 30 shared.
  pass 12, 21 shared.
  pass 13, 12 shared.
  pass 14, 6 shared.
  pass 15, 4 shared.
  pass 16, 2 shared.
  pass 17, 2 shared.
  pass 18, 1 shared.
  pass 19, 1 shared.
  pass 20, 1 shared.
[...snip...]
  pass 198, 1 shared.
  pass 199, 1 shared.
  pass 200, 1 shared.
fatal error: failed to route
make: *** [hardware.asc] Error 1
```

closed 2019-01-16 by gojimmypi

daveshah1, 2019-01-17:
FYI, this is a long known issue in the placement validity checks, where it will sometimes end up using more than 32 local inputs in a logic cell. It is properly fixed in [nextpnr](https://github.com/YosysHQ/nextpnr).

gojimmypi, 2019-01-18:
Is this issue documented somewhere? I spent a ton of time trying to figure it out. Certainly googling `"fatal error: failed to route" nextpnr` and related terms did not seem to help. Like most things - easy when you know the answer, eh! lol :)

As for nextpnr, I would have tried, but I've not yet been able to get it to compile as noted in https://github.com/YosysHQ/nextpnr/issues/214

Bazmundi, 2021-04-04:
Golly! Its so fixed, except I get it after 1) following [this install](http://www.clifford.at/icestorm/) and then 2) building from [this directory](url), see [the error dump of the following](https://pastebin.com/n0Dms6JU).

I have to also note that my third part directory (nextpnr/3rdparty/abseil-cpp) directory was empty and was causing a build fail.  So I  changed directory to my nextpnr/3rdparty and then cloned abeil-cpp using:

`git clone https://github.com/abseil/abseil-cpp.git`

Note, I have just installed fresh today and so cannot vouch for a fix in nextpnr, if that was supposed to remove the problem.

I am on Debian 10, running Python 3.7.3.

On the assumption the original store and having to fix that alone the way prompts the problem, I will need rip the lot out and try a fresh install of [symbiflow ](https://symbiflow.github.io/) instead.

@gojimmypi the compile problem is probably because the nextpnr distro is missing 3rd party dependencies.