YosysHQ/arachne-pnr #134: "src/util.hh:191:24: error: moving a local object in a return statement prevents copy elision"

tommythorn, 2019-02-24:
I'm building and testing as many things as I can in the 2019-02-21 Fedora/RISC-V build and hit a build error in arachne-pnr.  I haven't yet tried to fix it myself, but a quick look suggests it hasn't been reported before:

```
g++ -Isrc -std=c++11 -MD -O2 -Wall -Wshadow -Wsign-compare -Werror   -c -o src/line_parser.o src/line_parser.cc
In file included from src/bstream.hh:19,
                 from src/location.hh:19,
                 from src/chipdb.hh:19,
                 from src/chipdb.cc:16:
src/util.hh: In instantiation of ‘std::set<typename M::key_type> keys(const M&) [with M = std::map<CBit, bool>; typename M::key_type = CBit]’:
src/chipdb.cc:73:23:   required from here
src/util.hh:191:24: error: moving a local object in a return statement prevents copy elision [-Werror=pessimizing-move]
  191 |   return std::move(keys);
      |                        ^
src/util.hh:191:24: note: remove ‘std::move’ call
g++ -Isrc -std=c++11 -MD -O2 -Wall -Wshadow -Wsign-compare -Werror   -c -o src/pcf.o src/pcf.cc
```

g++: gcc version 9.0.1 20190219 (Red Hat 9.0.1-0.6) (GCC) 
arache-pnr: 840bdfdeb38809f9f6af4d89dd7b22959b176fdd
build command: make -C arachne-pnr -j8

tommythorn, 2019-02-24:
This is a temporary workaround:
```
make OPTDEBUGFLAGS='-MD -O2 -Wno-error=pessimizing-move'
```

kpfromer, 2019-07-15:
Yeah I am also getting this error trying to install arachne-pnr via the `arachne-pnr-git` arch AUR package.

JordiVM, 2019-07-28:
Same here, just tried to install `arachne-pnr` both through command line and arch AUR in Manjaro and had the same error on both.

daveshah1, 2019-07-28:
Have disabled -Werror, arachne-pnr is now essentially unmaintained so this seems like a reasonable enough solution.